LeClubSandwichStudio/LaCOOLBoard #3: Reformat JSON API (currently a mess)

3rwww1, 2018-02-06:
Currently, the JSON API has a lot of issues
- Measurements are not nested into a "sensors" JSON object, so it's hard to figure out an unknown configuration key from an unconfigured sensor.
- Similarly, configuration is not nested under a "config" object
- Boolean values are sometime returned as `true` or `false`, sometimes as `0` or `1`, **which is a big no-no**
- Embedded software version is not sent
- things in frenglsh (`actif` instead of `active`)

All in all, 
- It requires me to reparse the whole JSON to make cleanup the data and make sense of it, which is time wasted and frankly driving me nuts.
- On an embedded software upgrade, new application bugs can occur (and thats, like, really bad - consider hackers and makers we sell the COOL Board to). We should be able to display a message such as "your coolboard software is outdated, please upgrade"

We need to move to a software engineering perspective on doing things:
- make the API return what we have in the "Parsed JSON" section of the COOL Menu code drop-down
- document and publish the COOL Board API using a JSON Schema at least (http://json-schema.org/)
- make sure changes are discussed beforehand

3rwww1, 2018-03-01:
target format, as we discussed with @simon-lacool and @baltoche 

```json
{
  "static": {
    "hwVersion": "1.2.2",
    "macAddress": "60:01:94:1d:4c:b4",
    "ftdi": "...",
    "serialNumber": "CDC-2877-027"
  },
  "state": {
    "actuators": [
      false,
      false,
      false,
      false,
      false,
      false,
      false,
      false,
      false
    ],
    "system": {
      "fwVersion": "0.2.3",
      "timeSync": 1519814482,
      "manual": false,
      "publicIp": "178.20.51.116",
      "timestamp": "2018-03-01T10:56:33.000Z"
    },
    "sensors": {
      "wifiSignal": -76,
      "batteryVoltage": 0.01,
      "visibleLight": 289,
      "infraRed": 637,
      "ultraViolet": 0.45,
      "temperature": 24.22,
      "pressure": 992.38,
      "humidity": 36.45,
      "soilMoisture": 0,
      "ph": 7.56,
      "waterTemp": 24.56,
      "waterLevel": 5.4,
      "hydreon": 235456,
      "tipping": 245564,
      "wind": 3231321553,
      "ppfd": {
        "value": 6.7,
        "gain": 2
      },
      "waterMark": {
        "value": 6.3,
        "gain": 4
      },
      "color": {
        "temp": 4242,
        "r": 0.86,
        "g": 0.3,
        "b": 0.98,
        "c": 0.14
      }
    }
  },
  "config": {
    "led": {
      "ledActive": true
    },
    "clock": {
      "ntp": true,
      "timeServer": "europe.pool.ntp.org",
      "localPort": 2390
    },
    "general": {
      "logInterval": 300,
      "ireneActive": true,
      "jetpackActive": true,
      "externalSensorsActive": true,
      "serverTimeOut": 180,
      "sleepActive": false,
      "saveAsCsv": false
    },
    "sensors": {
      "builtIns": {
        "bme280": {
          "temperature": true,
          "humidity": true,
          "pressure": true
        },
        "st114X": {
          "visible": true,
          "ir": true,
          "uv": true
        },
        "batteryVoltage": true,
        "soilMoisture": true,
        "wifiSignal": true
      },
      "irene": {
        "waterTemp": {
          "active": true,
          "gain": 16
        },
        "ph": {
          "active": true,
          "gain": 16
        },
        "adc2": {
          "kind": "waterLevel",
          "active": false,
          "gain": true,
          "type": "voltage"
        },
        "calibration": {
          "pH7": 15092,
          "pH4": 20490,
          "step": -10.71064,
          "timestamp": "2018-02-27T09:31:42.000Z"
        }
      },
      "external": [
        {
          "reference": "CoolGauge",
          "type": "CoolGauge",
          "kind0": "tipping",
          "kind1": "hydreon",
          "kind2": "wind",
          "address": 0,
          "sensor3": "0",
          "kind3": "0"
        },
        {
          "reference": "NDIR_I2C",
          "type": "CO2",
          "connection": "I2C",
          "address": 77,
          "kind0": "0",
          "kind1": "0",
          "kind2": "0",
          "sensor3": "0",
          "kind3": "0"
        }
      ]
    },
    "actuators": [
      {
        "active": true,
        "inverted": true,
        "temporal": true,
        "low": [
          0,
          0,
          20,
          0
        ],
        "high": [
          0,
          0,
          6,
          0
        ],
        "type": {
          "primary": "hour"
        }
      },
      {
        "active": true,
        "inverted": true
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": false,
        "inverted": false
      },
      {
        "active": false,
        "inverted": false
      },
      {
        "active": false,
        "inverted": true
      }
    ],
    "wifi": {
      "nomad": 1
    },
    "mqtt": {
      "mqttServer": "52.17.46.139",
      "bufferSize": 3000,
      "inTopic": "$aws/things/6001941D4CB4/shadow/update/delta",
      "outTopic": "$aws/things/6001941D4CB4/shadow/update"
    }
  }
}
```

trap98, 2018-07-04:
For the configuration i think take json name is better

trap98, 2018-07-05:
First draft new single Json :
```
{
  "general": {
    "logInterval": 600,
    "ledActive":true,
    "ireneActive": true,
    "jetpackActive": true,
    "externalSensorsActive": true,
    "sleepActive": false,
    "mqttServer": "mqtts.lacoolboard.io"
  },
  "metrics": {
    "builtIns": {
      "BME280": {
        "temperature": true,
        "humidity": true,
        "pressure": true
      },
      "SI114X": {
        "visible": true,
        "ir": true,
        "uv": true
      },
      "vbat": true,
      "soilMoisture": true,
      "wallMoisture": false
    },
    "irene3000": {
      "waterTemp": {
		    "active":1,
		    "gain":16
	    },
	    "phProbe": {
		    "active":1,
		    "gain":16
	    },
	    "adc2": {
		    "active":0,
		    "gain":16,
		    "type":"voltage"
	    }
    },
    "externals": [
      {
        "reference":"SDS011",
        "type":"PM",
        "address":13,
        "kind0":"PM10",
        "kind1":"PM2.5"
      },
      {
        "reference":"NDIR_I2C",
        "type":"CO2",
        "address":77
      },
      {
        "reference": "SHT1x",
        "type": "SHT1x",
        "kind0": "SHT1x_Humidity",
        "kind1": "SHT1x_TempC"
      }
    ],
    "jetpack": [
      {
        "actif": false,
        "inverted": false,
        "temporal": false,
        "low": [0,5000,0,0],
        "high": [0,5000,0,0],
        "type": ["Temperature",""]
      },
      {
        "actif":true,
        "inverted":true,
        "temporal":true,
        "low":[0,0,20,0],
        "high":[0,0,7,0],
        "type":["","hour"]
      },
      {
        "actif":true,
        "inverted":true,
        "temporal":true,
        "low":[0,0,23,0],
        "high":[0,0,6,0],
        "type":["","hour"]
      },
      {
        "actif":true,
        "inverted":false,
        "temporal":true,
        "low":[0,600000,0,0],
        "high":[0,600000,0,0],
        "type":["",""]
      },
      {
        "actif":true,
        "inverted":false,
        "temporal":true,
        "low":[0,60000,0,0],
        "high":[0,60000,0,0],
        "type":["",""]
      },
      {
        "actif":true,
        "inverted":false,
        "temporal":true,
        "low":[0,120000,0,0],
        "high":[0,60000,0,0],
        "type":["",""]
      },
      {
        "actif":true,
        "inverted":false,
        "temporal":true,
        "low":[1500,0,6,0],
        "high":[1700,0,23,0],
        "type":["CO2","hour"]
      },
      {
        "actif":false,
        "inverted":false,
        "temporal":false,
        "low":[27,0,0,0],
        "high":[33,0,0,0],
        "type":["Temperature",""]
      },
      {
        "actif":false,
        "inverted":true,
        "temporal":false,
        "low":[0,0,6,0],
        "high":[0,0,18,0],
        "type":["","hour"]
      }
    ]
  }
}

```