LeClubSandwichStudio/LaCOOLBoard #4: Sending external sensors as nested JSON object

3rwww1, 2018-02-06:
for instance we send the values of externel sensors in the string. example with a CSS811:
`..."CO2":768,"VOCT":54,"temp":25...`
the configuration is : 
```
"sensor0":
	{
		"reference":"Adafruit_CCS811",
		"type":"AirQuality",
		"address":90,
		"kind0":"CO2",
		"kind1":"TVOC",
		"kind2":"temp"
	}
```
what we want to get is the type as key for the nested array, like this:
`..."AirQuality":{"CO2":768,"VOTC":54,"temp":25}...`

3rwww1, 2018-03-01:
// 20180301131130
// file:///home/erwan/code/spec.json

{
  "static": {
    "hwVersion": "1.2.2",
    "macAddress": "60:01:94:1d:4c:b4",
    "ftdi": "...",
    "serialNumber": "CDC-2877-027"
  },
  "state": {
    "actuators": [
      false,
      false,
      false,
      false,
      false,
      false,
      false,
      false,
      false
    ],
    "system": {
      "fwVersion": "0.2.3",
      "timeSync": 1519814482,
      "manual": false,
      "publicIp": "178.20.51.116",
      "timestamp": "2018-03-01T10:56:33.000Z"
    },
    "sensors": {
      "wifiSignal": -76,
      "batteryVoltage": 0.01,
      "visibleLight": 289,
      "infraRed": 637,
      "ultraViolet": 0.45,
      "temperature": 24.22,
      "pressure": 992.38,
      "humidity": 36.45,
      "soilMoisture": 0,
      "ph": 7.56,
      "waterTemp": 24.56,
      "waterLevel": 5.4,
      "hydreon": 235456,
      "tipping": 245564,
      "wind": 3231321553,
      "ppfd": {
        "value": 6.7,
        "gain": 2
      },
      "waterMark": {
        "value": 6.3,
        "gain": 4
      },
      "color": {
        "temp": 4242,
        "r": 0.86,
        "g": 0.3,
        "b": 0.98,
        "c": 0.14
      }
    }
  },
  "config": {
    "led": {
      "ledActive": true
    },
    "clock": {
      "ntp": true,
      "timeServer": "europe.pool.ntp.org",
      "localPort": 2390
    },
    "general": {
      "logInterval": 300,
      "ireneActive": true,
      "jetpackActive": true,
      "externalSensorsActive": true,
      "serverTimeOut": 180,
      "sleepActive": false,
      "saveAsCsv": false
    },
    "sensors": {
      "builtIns": {
        "bme280": {
          "temperature": true,
          "humidity": true,
          "pressure": true
        },
        "st114X": {
          "visible": true,
          "ir": true,
          "uv": true
        },
        "batteryVoltage": true,
        "soilMoisture": true,
        "wifiSignal": true
      },
      "irene": {
        "waterTemp": {
          "active": true,
          "gain": 16
        },
        "ph": {
          "active": true,
          "gain": 16
        },
        "adc2": {
          "kind": "waterLevel",
          "active": false,
          "gain": true,
          "type": "voltage"
        },
        "calibration": {
          "pH7": 15092,
          "pH4": 20490,
          "step": -10.71064,
          "timestamp": "2018-02-27T09:31:42.000Z"
        }
      },
      "external": [
        {
          "reference": "CoolGauge",
          "type": "CoolGauge",
          "kind0": "tipping",
          "kind1": "hydreon",
          "kind2": "wind",
          "address": 0,
          "sensor3": "0",
          "kind3": "0"
        },
        {
          "reference": "NDIR_I2C",
          "type": "CO2",
          "connection": "I2C",
          "address": 77,
          "kind0": "0",
          "kind1": "0",
          "kind2": "0",
          "sensor3": "0",
          "kind3": "0"
        }
      ]
    },
    "actuators": [
      {
        "active": true,
        "inverted": true,
        "temporal": true,
        "low": [
          0,
          0,
          20,
          0
        ],
        "high": [
          0,
          0,
          6,
          0
        ],
        "type": {
          "primary": "hour"
        }
      },
      {
        "active": true,
        "inverted": true
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": true,
        "inverted": false
      },
      {
        "active": false,
        "inverted": false
      },
      {
        "active": false,
        "inverted": false
      },
      {
        "active": false,
        "inverted": true
      }
    ],
    "wifi": {
      "nomad": 1
    },
    "mqtt": {
      "mqttServer": "52.17.46.139",
      "bufferSize": 3000,
      "inTopic": "$aws/things/6001941D4CB4/shadow/update/delta",
      "outTopic": "$aws/things/6001941D4CB4/shadow/update"
    }
  }
}

simon-lacool, 2018-05-09:
https://github.com/LaCoolCo/LaCOOLBoard/commit/acbb2c306e71c093b3c24a30fd846ca86b8eb1e3

closed 2018-06-07 by 3rwww1