LeClubSandwichStudio/LaCOOLBoard #14: Choose a style-guide (from clang-format)

3rwww1, 2018-02-15:
I think it's time to choose a style guide for C++ code

Prerequisite : 
- don't choose our own, choose one from some big corp or project so we don't have to maintain it/troll around the issue
- choose one that has [clang-format support] (https://clang.llvm.org/docs/ClangFormatStyleOptions.html)

The possible styles are 
* LLVM 
* Google 
* Chromium 
* Mozilla
* WebKit

For the moment, Simon and myself are favoring Mozilla's. If one of you have a good case to make against that choice, please tell us here!

In max 1 week time, we'll reformat the whole code base to the chosen style, so hurry up if you have something to say !

PS: I know @ashiroji favors [Allman](https://en.wikipedia.org/wiki/Indentation_style#Allman_style) but no big player in the industry is actually using it, so no support for clang-format, so not an option, sorry!

ashiroji, 2018-02-15:
I had a quick look on the possible styles, Mozilla is fine by me!

3rwww1, 2018-02-17:
Ok, so I've tried formatting everything Mozilla style and encountered a weird formatting rule for methods. Mozilla style enforce putting the return type of a method on its own separate line, like this:
```
int
CoolBoard::isConnected()
{
```
instead of a oneliner like
`int CoolBoard::isConnected() {`
So to avoid this I chose to stick with the LLVM style guide for my pull-request, unless you feel otherwise and absolutely want Mozilla style

closed 2018-02-20 by 3rwww1