LeClubSandwichStudio/LaCOOLBoard #27: Can't add a external sensor OTA

simon-lacool, 2018-03-01:
I tried to add a external sensor (NDIR CO2 sensor) over the air and it failed... the I tried all kinds of figures, sending just the config for new sensor, sending everything. 
The configuration is arriving partially in the configuration and the new sensor didn't work. after flashing the howl stuff tu the SPIFFS everything worked nice.

simon-lacool, 2018-03-22:
In fact the problem is slightly more global than that. It has to do with the REST API and the fact that we only get the difference between reported and desired. 
Example:
_shadow state:_
`{"val1"=1, "val2"=2, "val3"=3}`
_json from user:_
`{"val1"=2, "val2"=2, "val3"=1}`
_json in desired channel_
`{"val1"=2, "val3"=1}`
_json in memory after update_
`{"val1"=2, "val3"=1}`

the REST API acts as intended but the coolboard **does not keep unchanged values**. "that's all"...

3rwww1, 2018-06-07:
To review / define