LeClubSandwichStudio/LaCOOLBoard #41: Upgrade to SDK 2.4.1

3rwww1, 2018-04-16:


trap98, 2018-04-16:
Problem at the compilation with SDK 2.4.1 on the library DS1337RTC.

Kameeno, 2018-04-20:
on the new porting on the last SDK, cause bug about connectivity, so, if on your spiffs you got some saved SSID/PASSWORD on wificonfig.json the board, i think she did not take the credential and arent compared with the current wifi arround.

ok, if you flash your firmware and on your spiffs, wificonfig.json did not contain some wifi credentials, you arrive to connect, i do my OTA update, (with the same firmware) on reboot she didn't getting credentials from wificonfig.json

actually i missed my coolboard to make a good bugfix about

closed 2018-06-07 by 3rwww1