LeClubSandwichStudio/LaCOOLBoard #43: Over-the-air self updates

3rwww1, 2018-04-17:


3rwww1, 2018-04-17:
@trap98 @Kameeno I want you both to collaborate on this and make a clean pull request by the end of the week. @Kameeno please bring @trap98 up-to-speed on the way the current firmware/API works and ask for my help... When you **really** need it :)

Kameeno, 2018-04-19:
it's going good for the classic HTTP update, so now..
i'm looking for https updates, i found this on the ESP8266HTTPUpdate.cpp

> HTTPUpdateResult ESP8266HTTPUpdate::update(const String& host, uint16_t port, const String& uri, const String& currentVersion,
>         bool https, const String& httpsFingerprint, bool reboot)
> {
> 

the reference say this : 

> **t_httpUpdate_return ESP8266HTTPUpdate::update** (
>   |   | String | host,
>   |   | uint16_t | port,
>   |   | String | url = "/",
>   |   | String | current_version = "",
>   |   | bool | https = false,
>   |   | String | httpsFingerprint = "",
>   |   | bool | reboot = true
>   | )

the document of reference is here : 
[https://links2004.github.io/Arduino/df/d31/class_e_s_p8266_h_t_t_p_update.html](url)
i propose to have host reference (where you want) and to send the address of the binary file,
for the rest... **i found a LOOP of update, the cause is from the mqtt ( or not?)
so, when i send my json message, the coolboard she getting on update and she reboot with new firmware.. and she re-check on the mqtt but... she found again the last message about OTA.** 

one more thing

the function on the original library she do this:

> HTTPUpdateResult ESP8266HTTPUpdate::handleUpdate(HTTPClient& http, const String& currentVersion, bool spiffs)
> {
> 
>     HTTPUpdateResult ret = HTTP_UPDATE_FAILED;
> 
>     // use HTTP/1.0 for update since the update handler not support any transfer Encoding
>     http.useHTTP10(true);
>     http.setTimeout(8000);
>     http.setUserAgent(F("ESP8266-http-Update"));
>     http.addHeader(F("x-ESP8266-STA-MAC"), WiFi.macAddress());
>     http.addHeader(F("x-ESP8266-AP-MAC"), WiFi.softAPmacAddress());
>     http.addHeader(F("x-ESP8266-free-space"), String(ESP.getFreeSketchSpace()));
>     http.addHeader(F("x-ESP8266-sketch-size"), String(ESP.getSketchSize()));
>     http.addHeader(F("x-ESP8266-sketch-md5"), String(ESP.getSketchMD5()));
>     http.addHeader(F("x-ESP8266-chip-size"), String(ESP.getFlashChipRealSize()));
>     http.addHeader(F("x-ESP8266-sdk-version"), ESP.getSdkVersion());

are the http header the cause of the block to make a http/https REQUEST to GITHUB?
and i'm little bit confused about "currentVersion" if is meaning "HTTP/1.0" version or..... just the current version to send a HTTP Post of the firmware running now ?

Kameeno, 2018-04-20:
on the new porting on the last SDK, cause bug about connectivity, so, if on your spiffs you got some saved SSID/PASSWORD on wificonfig.json the board, i think she did not take the credential and arent compared with the current wifi arround.

ok, if you flash your firmware and on your spiffs, wificonfig.json did not contain some wifi credentials, you arrive to connect, i do my OTA update, (with the same firmware) on reboot she didn't getting credentials from wificonfig.json

actually i missed my coolboard to make a good bugfix about

PR #68 (2018-06-08): Add HTTPS OTA support
Resolves: #43
closed 2018-06-22 by 3rwww1