LeClubSandwichStudio/LaCOOLBoard #55: logInterval() can't be set higher than  4294 seconds

simon-lacool, 2018-05-14:
for instance we logged once every hour or shorter, now we head a client that want's one log per hour. So I put 86400 seconds as logInterval but the card reboots every ~10 minutes..
I think our problem is related to this :
https://www.esp8266.com/viewtopic.php?f=32&t=7867
and also to this merge : 
https://github.com/esp8266/Arduino/pull/4234

simon-lacool, 2018-05-14:
also discussed here:
https://electronics.stackexchange.com/questions/306374/extend-esp8266-12-f-deep-sleep-time?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa

https://www.esp8266.com/viewtopic.php?f=6&t=12908

3rwww1, 2018-06-07:
Test with `logInterval` 12h - 24h

trap98, 2018-06-08:
Not functionnal, with 12 and 24 hours. Sometimes the board go to sleep with white led on and ap on

Kameeno, 2018-06-18:
```//this calculation was taken verbatim from the SDK api reference for SDK 2.1.0.
//Note: system_rtc_clock_cali_proc() returns a uint32_t, even though system_deep_sleep() takes a uint64_t.
uint64_t EspClass::deepSleepMax()
{
  //cali*(2^31-1)/(2^12)
  return (uint64_t)system_rtc_clock_cali_proc()*(0x80000000-1)/(0x1000);

}``` 
someone has checked esp.h and esp.cpp ?

Axe-R, 2018-06-28:
- I made a test with a loginterval of 4 hours, it did a measure 3h30 later and then nothing more. 

- Another test with a loginterval of 5 hours, it didn't do anything and when I reset the coolboard, it has a white led on.