LeClubSandwichStudio/LaCOOLBoard #77: Prototype a REST API using ESP8266 http server

3rwww1, 2018-06-07:


Kameeno, 2018-06-28:
is done, is available on: http://192.168.4.1/scan
he get an array like that: 
```
[
  {
    "rssi": -85,
    "ssid": "network0",
    "bssid": "00:00:00:00:00:00",
    "channel": 6,
    "secure": 8,
    "hidden": false
  },
  {
    "rssi": -47,
    "ssid": "network1",
    "bssid": "00:00:00:00:00:00",
    "channel": 6,
    "secure": 8,
    "hidden": false
  },
  {
    "rssi": -77,
    "ssid": "network2",
    "bssid": "00:00:00:00:00:00",
    "channel": 6,
    "secure": 8,
    "hidden": false
  },
  {
    "rssi": -68,
    "ssid": "network3",
    "bssid": "00:00:00:00:00:00",
    "channel": 11,
    "secure": 8,
    "hidden": false
  },
  {
    "rssi": -53,
    "ssid": "network4",
    "bssid": "00:00:00:00:00:00",
    "channel": 11,
    "secure": 7,
    "hidden": false
  },
  {
    "rssi": -71,
    "ssid": "network5",
    "bssid": "00:00:00:00:00:00",
    "channel": 11,
    "secure": 8,
    "hidden": false
  },
  {
    "rssi": -77,
    "ssid": "network6",
    "bssid": "00:00:00:00:00:00",
    "channel": 11,
    "secure": 8,
    "hidden": false
  }
]

```

nlsdvl, 2018-07-16:
the task related to network service publishing/discovery could be split in another ticket (is it already ?)
I've been testing it today, the discovery worked with:
-- `gssdp-discovery -m all`
-- `dns-sd -B` 
I was able to access the device on the network with `http://coolboard-2c3ae84fbe5b.local/` at some point, but this did not work reliably at later attempts.

I am now looking for an appropriate package that can list the service in javascript.

regarding the API, I was able to use the following endpoints with Curl : 
/add/wifi
/reset/wifi
/wifi/scan

the API calls did not work with Postman however.

nlsdvl, 2018-07-16:
API calls from JS working fine.

nlsdvl, 2018-07-16:
service discovery working great with:  https://github.com/taoyuan/nbonjour

nlsdvl, 2018-07-17:
> request: POST /wifi/saved

There is no way to validate the password that the user posts without disconnecting. So a success response means the password is stored successfully. But not that it is a valid password for that network.

nlsdvl, 2018-07-18:
How about PUT & DELETE ? We need to be able to update or delete WiFis ... current @Kameeno implemented a resetWiFi method for that purpose.

nlsdvl, 2018-07-18:
Maybe not a priority, but there's no mention of a `/disconnect` command. I expect `/connect` implicitly disconnect if needed. Disconnect would be used to get back to AP mode, or for test/debug purpose.

cc @3rwww1 @Kameeno  you guys reading this ?

nlsdvl, 2018-07-18:
I'd expect an error (4xx client error) when asking the board to forget to a Wifi but the board being actually connected to the same wifi. Correct ?

Kameeno, 2018-07-19:
@nlsdvl  yes, i'm not arriving to reconnect to another wifiNetwork without crashing...