LeClubSandwichStudio/LaCOOLBoard #87: SHT01 temperature and soil humidity sensor

3rwww1, 2018-06-11:


Kameeno, 2018-06-11:
is not possible to make it works on i2c pins, he cause wrong read values of i2c sensors, the functions like ```wire.end();``` are only reserved for AVR mcu's, so for the moment is running on pin's 12 and 13 with a pullup on 12 (data pin) & a pulldown for pin 13 (clock pin),

these pins are on the board for the analog switch(12) and (13) to enable the onboard Moisture sensor. 

putting this sensor to i2c pins caused to me strange stuff like, mqtts disconnection (without possibility to reconnect, NTP server unresponsive after running it for 15/20 minutes) .

3rwww1, 2018-06-11:
@Kameeno, regarding your latest comment, have you seen with @simon-lacool what would be the most cost-effective solution ? Use the soil moisture fork pins and not our I²C pins ?

closed 2018-06-22 by 3rwww1