LeClubSandwichStudio/LaCOOLBoard #141: Reliable way to make SHT1x work, with jetpack

3rwww1, 2018-06-20:


Kameeno, 2018-06-26:
we got another sht? so normally can work, we have changed the pin to use. 
if i found one i go to test it

simon-lacool, 2018-06-27:
pins where changes in commit https://github.com/LaCoolCo/LaCOOLBoard/commit/9f11a2a1ea9b84aea8802ff6e58be2fff29f1025 . 
one SDS011 and one SHT01 are on my table!

simon-lacool, 2018-06-27:
Pin assignment on 0 & 12 is not working with the jetpack. Please deprecate this way to get this sensor if you use the Jetpack!
**It blocks the CoolBoard like if you put it in load position.**

Using this sensor without jetpack seems to work nice.

simon-lacool, 2018-06-29:
after 3 days without problem I confirm that pin 0 and 12 work fine for the SHT1x on the CoolCave with all other sensors: CO2, SDS011, onboard sensors...
I close this issue, but anyway I will have an eye on it...

closed 2018-06-29 by simon-lacool