LeClubSandwichStudio/LaCOOLBoard #148: Crash during OTA update

3rwww1, 2018-06-22:
**Steps to reproduce**

* plug a COOL Board in
* send an MQTT configuration update

**Expected**
* New configuration is applied
* Changes can be seen in the COOL Menu

**Actual**
* COOL Board crashes and reset endlessly

Kameeno, 2018-06-23:
on the last master it's looking good all, i was made a lot of changes of my configuration via mqtt, like change interval log, enable the wallmoisture flag or change the deepsleep flag, it looking pretty good
i tired also to add a new wifi, or new externalsensors also. no crash for the moment

Kameeno, 2018-06-24:
probably found a restartloop, for the rest, if we stay with: 
```{
  "state": {
    "desired": {
      "externalSensors": {
        "sensorsNumber": 1,
        "sensor0": {
          "reference": "SHT1x",
          "type": "SHT1x",
          "kind0": "SHT1x_Humidity",
          "kind1": "SHT1x_TempC",
          "address": 0
        }
      }
    }
  }```
only SensorNumber get the update, 
on the debug message i cant find the jsonObject of 'sensor0'

but if i rename ```externalSensors``` on ```updateExternalSensors``` i can get with succes the various "sensor0,sensor1,sensor2" objects.

ps if we want to update config from ```CoolFileSystem::updateConfigFiles(stateDesired);```
need a update from: 
```static constexpr ConfigFile CONFIG_FILES[] = {
      {"CoolBoard", "/coolBoardConfig.json"},
      {"CoolSensorsBoard", "/coolBoardSensorsConfig.json"},
      {"CoolBoardActor", "/coolBoardActorConfig.json"},
      {"rtc", "/rtcConfig.json"},
      {"led", "/coolBoardLedConfig.json"},
      {"jetPack", "/jetPackConfig.json"},
      {"irene3000", "/irene3000Config.json"},
      {"mqtt", "/mqttConfig.json"},
      {"wifi", "/wifiConfig.json"}};

static const uint8_t CONFIG_FILES_COUNT = 9;

void CoolFileSystem::updateConfigFiles(JsonObject &root) {
  for (uint8_t i = 0; i < CONFIG_FILES_COUNT; ++i) {
    JsonObject &json = root[CONFIG_FILES[i].code];
    if (json.success()) {
      CoolFileSystem::fileUpdate(json, CONFIG_FILES[i].path);
    }
  }
}```
to:
```static constexpr ConfigFile CONFIG_FILES[] = {
      {"CoolBoard", "/coolBoardConfig.json"},
      {"CoolSensorsBoard", "/coolBoardSensorsConfig.json"},
      {"CoolBoardActor", "/coolBoardActorConfig.json"},
      {"externalSensors", "/externalSensorsConfig.json"},
      {"rtc", "/rtcConfig.json"},
      {"led", "/coolBoardLedConfig.json"},
      {"jetPack", "/jetPackConfig.json"},
      {"irene3000", "/irene3000Config.json"},
      {"mqtt", "/mqttConfig.json"},
      {"wifi", "/wifiConfig.json"}};

void CoolFileSystem::updateConfigFiles(JsonObject &root) {
  for (uint8_t i = 0; i < sizeof(CONFIG_FILES); ++i) {
    JsonObject &json = root[CONFIG_FILES[i].code];
    if (json.success()) {
      CoolFileSystem::fileUpdate(json, CONFIG_FILES[i].path);
    }
  }
}```

on CONFIG_FILES[] we can't update ```/externalSensorsConfig.json```
p.s. why we need to declare the size of our array?

Kameeno, 2018-06-24:
need a update from: 
```
static constexpr ConfigFile CONFIG_FILES[] = {
      {"CoolBoard", "/coolBoardConfig.json"},
      {"CoolSensorsBoard", "/coolBoardSensorsConfig.json"},
      {"CoolBoardActor", "/coolBoardActorConfig.json"},
      {"rtc", "/rtcConfig.json"},
      {"led", "/coolBoardLedConfig.json"},
      {"jetPack", "/jetPackConfig.json"},
      {"irene3000", "/irene3000Config.json"},
      {"mqtt", "/mqttConfig.json"},
      {"wifi", "/wifiConfig.json"}};

static const uint8_t CONFIG_FILES_COUNT = 9;

void CoolFileSystem::updateConfigFiles(JsonObject &root) {
  for (uint8_t i = 0; i < CONFIG_FILES_COUNT; ++i) {
    JsonObject &json = root[CONFIG_FILES[i].code];
    if (json.success()) {
      CoolFileSystem::fileUpdate(json, CONFIG_FILES[i].path);
    }
  }
}
```
to:
```
static constexpr ConfigFile CONFIG_FILES[] = {
      {"CoolBoard", "/coolBoardConfig.json"},
      {"CoolSensorsBoard", "/coolBoardSensorsConfig.json"},
      {"CoolBoardActor", "/coolBoardActorConfig.json"},
      {"externalSensors", "/externalSensorsConfig.json"},
      {"rtc", "/rtcConfig.json"},
      {"led", "/coolBoardLedConfig.json"},
      {"jetPack", "/jetPackConfig.json"},
      {"irene3000", "/irene3000Config.json"},
      {"mqtt", "/mqttConfig.json"},
      {"wifi", "/wifiConfig.json"}};

void CoolFileSystem::updateConfigFiles(JsonObject &root) {
  for (uint8_t i = 0; i < sizeof(CONFIG_FILES); ++i) {
    JsonObject &json = root[CONFIG_FILES[i].code];
    if (json.success()) {
      CoolFileSystem::fileUpdate(json, CONFIG_FILES[i].path);
    }
  }
}
```
on CONFIG_FILES[] we can't update ```/externalSensorsConfig.json```
p.s. why we need to declare the size of our array?

Kameeno, 2018-06-24:
i'm on test, it's running good for now

Kameeno, 2018-06-24:
ok, using a sizeof() is causing a crash, i cant arrive to end my function.
for the rest to make it easy what i do is: 
```
{  
   "state":{  
      "desired":{  
         "externalSensors":{  
            "sensorsNumber":1,
            "sensor0":{  
               "reference":"SHT1x",
               "type":"SHT1x",
               "kind0":"SHT1x_Humidity",
               "kind1":"SHT1x_TempC",
               "address":0,
               "kind2":"0",
               "kind3":"0"
            }
         }
      },
      "reported":{  
         "externalSensors":null
      }
   }
}
``` 
to wipe the old configuration, on the shadow of mqtt, 
when i receive it, i verify if is the argument match with ```externalSensors``` and i rewrite the file inside the spiffs with the entire String of my desired state of externalSensors

closed 2018-06-27 by 3rwww1