LeClubSandwichStudio/LaCOOLBoard #162: Need ESP.restart() if coolboard tried x number of connection mqtts do a restart. (or reset)

Kameeno, 2018-06-28:
So, from a lot of time on firmware we arrive to a rare cases when CoolBoard (on onlineMode) retry to connect without success and the only solution i found is to do a manual reset pushing the button.

3rwww1, 2018-06-28:
@Kameeno why is this a blocker/high priority ? 

Are messages saved on SPIFFS ? If yes, no need to worry, they'll be replayed.

Do you have an example of that hapenning in online mode ?

Could you copy logs here ?

Kameeno, 2018-06-28:
she just retry for hours and hours to reconnect, and she save her logs when she need to do it on spiffs, 
coolcave for examples she was offline from yesterday up to today, also my board at home.
we can talk about with @simon-lacool 
the same thing was happened with the "external station" 4 days ago

simon-lacool, 2018-07-10:
I think this problem is not finished since I observe the CoolCave hanging itself several times since our wifi hotspot had some problems.. 
I put the issue back to in testing. @Axe-R @gui4d @trap98 take care to install a Rpi on the coolcave to finally catch the moment when it crashes..