LeClubSandwichStudio/LaCOOLBoard #177: Adafruit Lux Sensor

3rwww1, 2018-07-04:
**Adafruit Lux sensor**
Implement Adafruit Lux sensor !

Kameeno, 2018-07-05:
i'm looking on this, what you think about @simon-lacool  ?
https://www.maximintegrated.com/en/products/sensors/MAX44009.html
there's a lot of module on online stores about

simon-lacool, 2018-07-09:
I taught about this sensor:
https://www.adafruit.com/product/1980
but yes the MAX44009 has a higher dynamic range and BoB's are really cheap, let's buy one and see if it's stable..
here I found a interesting paper about these two sensors:
https://ac.els-cdn.com/S1474667015373596/1-s2.0-S1474667015373596-main.pdf?_tid=acf56ab8-e63a-4ff4-ad02-98dd6656ea98&acdnat=1531132958_b3526cc7bd0be4d94262f83ccbfbcd19