LeClubSandwichStudio/LaCOOLBoard #193: Wrong link of firmware in OTA update

Axe-R, 2018-08-07:
**Describe the bug**
When I update the firmware of my CoolBoard with a wrong link, then the CoolBoard tries over and over again to update it. It doesn't measure anything and the LED is blinking orange.

**Steps to reproduce**
Steps to reproduce the behavior:
1. Send an update of the firmware version with a wrong link
2. Wait for the listening message

**Expected behavior**
The CoolBoard is trying several times to update its firmware & then keeps its actual version of the firmware.

**Actual behavior**
The CoolBoard is trying to update the firmware over and over again.

Kameeno, 2018-08-08:
in truth, when is starting to make (any wrong or right) firmware update, in case of fail when he reboot he start her normal work, after, is going to re-find on the the MQTT Shadow the firmware update message.
i thing can be right to do not just a counter or a fix, but a small library called "ErrorReason" to log inside a json file some errors we can arrive to have, for example: 
the wrong reading of a sensor, installed on the config but unreachable, 
problem with some wifi networks when we start to have, 
the problems of the firmware update.
and at the fail we reboot and we publish the message.

what do you think about @Axe-R @3rwww1 @simon-lacool @trap98  ?

Kameeno, 2018-08-08:
```
  "errorLog": {
    "2018-08-08T12:43:26Z":{
      "wifi": {
        "wifiState": 3,
        "mqttsState": "DISCONNECTED"
      },
      "firmwareUpdate": {
        "lastError": "UPDATE_ERROR_MAGIC_BYTE"
        },
      "Sensor": {
        "Adafruit_CCS811": "UNREACHABLE"
      }
    }
  }
```